devDucks/astroarch #121: All is working for me on PI 5 8gb but...

rapcanin, 2024-11-02:
VNC is awful...
Not usable at all.
I have used via wifi direct or via my modem, signal is full but it lags like nothing I have seen. Like the pc is pentium 365.
I have lost a good night because of this. And started to believe that astrophotography ia not possible using thr pi 5.
But let me see if anything can be changed or if someone else have a solution.
Thanks in advance!
P.s.
I have bought pi 5 because of this, I want to break zwo monopoly and to be able to use my Touptek camera with my Hyperstar

FireNWater, 2024-11-02:
I'm using a Pi-5 quite successfully with Astro Arch. . . it works.  We'll get you going. .

rapcanin, 2024-11-02:
Yes, it works with all my equipment, but VNC is crazy leggy and buggy.
I tried via web browser, and via more then two VNC viewers, used option for graphics and displays to medium, even on minimum it doesn't work and it is not usable.

kamilczas, 2024-11-03:
I'm using TigerVNC is the best for me zero delays, but my RP5 is connected direct to my switch in my room throught ethernet cable, so i can use my rp on my all devices in LAN. When i'm using rp wireless it was so laggy

BenoitBotton, 2024-11-07:
for me, on wifi, TigerVNC and another client (can't remember which, I am not on the right system right now) will virtually never connect. when it occasionally does, it is very slow. so I use the browser which is useable.
I wonder whether wifi signal is a bit weak on the pi. might we worse now I just added an M.2 hat and nvme SSD (in teh process of setting up). I will try a USB wifi dongle if I can get my hand the one I have somewhere

rapcanin, 2024-11-07:
Like this, I think it is not usable wirelessly

FireNWater, 2024-11-07:
Have you modded your Rpi-5 for external WiFi antenna? There are many YT videos, print you don’t have the skills / equipment there are folks offering the service on Etsy.

BenoitBotton, 2024-11-08:
I did not know of this mod. I will investigate. thanks

FireNWater, 2024-11-08:
I have the equipment and parts to do this mod.  I'm on the AA Discord channel - send me a DM if you need it done.

rapcanin, 2024-11-08:
> I have the equipment and parts to do this mod. I'm on the AA Discord channel - send me a DM if you need it done.

Are you based in EU?

FireNWater, 2024-11-08:
No, US.

rapcanin, 2024-11-08:
Since I have minipc maybe I will think how to munt my mini pc on my ring and start with nina, or I could try installing antenna to my pi5 and give a second try to AA or something else that can work on pi5

sc74, 2024-11-08:
Are you using AstroArch with the original settings or have you changed something like kde compositor for the display?

rapcanin, 2024-11-08:
No, it was all original, and before using it I have done all available updates

sc74, 2024-11-08:
Could you do this order, right after vnc instability

systemctl status x0vncserver

sc74, 2024-11-08:
What type of PC/Mac, Windows/Linux/macOS OS and which VNC client are you connecting with?

sc74, 2024-11-09:
Could you give us the result of the command iw wlan0 info | grep channel?

Jimbolaya64, 2025-03-28:
I have found that changing the background to a solid color greatly improves VNC performance. Still slightly laggy when on wi-fi, but very usable.

MattBlack85, 2025-10-30:
this issue is old, but the newest version has also rdp available  that may solve the vnc issues