devDucks/astroarch #124: Slow camera connection over USB3 bus

onley, 2024-11-18:
Very nice work, the whole distro works beautifully. I do however have a problem that I have seen a couple others report in my 2 days of Googling but I cannot find any fixes that work. I have an ZWO ASI183MC and on my laptop I can run full res at 10-15 FPS but on the 8GB PI4 I cannot get better than .9 FPS, in the grand scheme of things with long exposures not a big problem but trying to focus the rig, plate solve etc the speed can be a problem. I know this is not an Astroarch problem but since you know the guts of the distro I thought you might have some insight on how to improve the situation. Interestingly the ZWO is just as fast on USB2 as USB3. There is a 'quirks' fix floating around for slow USB3 external drives using UAS but the fix does nothing for the ZWO.

sc74, 2024-11-19:
Hi. Could you clarify what you are comparing? Under Windows what is your software to obtain your 10-15fps? And what are the settings? Binning to 1? What are the characteristics of your telescope?

For USB 2 or 3 the real difference is notable in lucky imaging. It may be interesting to see if this is not coming from your hard drive with the command hdparm -t /dev/sda

onley, 2024-11-19:
I run linux on my laptop and before Astroarch had been using AstroDMX so that is what I am comparing on both the laptop and PI. Binning is 1x1 and I see the problem with or without the harddrive (Harddrive is only for images Asrtoarch is running from sd card). There is no issue with imaging other than waiting for each image to download which as I mentioned is insignificant during imaging but quite time consuming during operations such as focusing and solving. As I mentioned the slow USB3 data rate is a known issue with the PI4 and there is a quirks fix for UAS harddrives/sdcards but it does not help the camera. At full res I am seeing download times of 3 seconds / image. I don't know if it is fixable but wondered if you might have a suggestion as you are obviously much more familiar with the guts of Astroarch and might be able to identify a USB driver issue.

FireNWater, 2024-11-19:
Have you tried adjusting the USB buffer sizes in the ONdI driver?

onley, 2024-11-19:
I upped the driver usb buffer to 1024M and even set the upload folder to a ramdrive and down load time for full frame image is still 2.8 seconds.

sc74, 2024-11-19:
It seems to me that your difference in fps comes from using an sd card for the pi and probably an ssd for your laptop. The SD card greatly limits the speed of software execution and data exchange between the cam and the disks. Try an SSD or NVMe instead of your SD card

onley, 2024-11-19:
The harddrive referenced earlier is a USB3 ssd and I set the upload folder from the indi driver for the camera to a ramdisk.  are you suggesting running Astroarch from the SSD?

sc74, 2024-11-19:
Yes, that's what using an SSD to run AstroArch

onley, 2024-11-20:
I've imaged the ssd to boot astroarch but the pi will not boot. I can boot from the sd card and mount the ssd partitions 1 and 2 and read the file system but am stuck.

onley, 2024-11-20:
I re-imaged on the pi itself and it now boots from SSD on USB3. Camera does 0.1 FPS in AstroDMX at 10ms exposure full res.

sc74, 2024-11-20:
Your SSD problem is strange. If you copy the image to your ssd and plug it into the top blue usb3 port on your pi without the sd card, you should boot without difficulty. 

Is your camera plugged into the second lower USB3 port? Could you give us the result of the lsusb -tv command?

onley, 2024-11-20:
There was a problem with the first image, I re-imaged the ssd and it booted fine as I stated above. However while I can write 0's to the drive with dd at 591MB/sec I can only get 0.1 FPS from the camera. I am currently imaging an sdcard with Raspbian to see if the problem is present there as well.

onley, 2024-11-21:
The Frame Rate is less than 1 FPS on Raspbian as well. Either the Pi4 USB3 implementation is a train wreck or there is a problem with the ZWO drivers and compatibility with the Pi4. As I said earlier AstroDMx on my Debian Bookworm laptop runs about 10FPS on USB3 while Raspbian Bookworm on the Pi4 gets less than 1FPS on USB3. At this point I think I am going to be using my laptop instead of a Pi4 which makes me quite unhappy.

sc74, 2024-11-21:
I understand. It is true that with a Pi, you cannot obtain the same responsiveness as with a PC. Several hardware-related bottlenecks explain this, as does the price. For example, the pi will never be a PC to do planetary work at 100 FPS. But it seems to me that you can get at least 20 FPS with FireCapture.
Now it is possible that one can improve the FPS rate with the help of AstroDMX user and some tweaks

onley, 2024-11-21:
The Pi5 has twice the processor speed and claims an improved USB3 bus improvement. I have ordered on and am hoping I am not disappointed. I found a thread on a forum where someone frustrated with this problem wrote a small utility to do nothing but download a frame from a ZWO camera into ram, he was only able to get a couple FPS. I didn't expect the same performance as my laptop but tasks such as focus are mind numbingly painful at 2 seconds per frame. I'll post back after I have Astroarch installed on the Pi5 to let you know how it goes.

LuckyEddie47, 2024-11-21:
I have run Astroarch (and Raspberry Pi OS & StellarMate) on a Pi5 with a ZWO. Using an SSD I'd estimate is at around x5 the performance of a Pi4. I haven't made any measurements at full frame.

onley, 2024-11-25:
Well, I tried the Pi5 and a miniPC. The Pi5 is as LuckyEddie47 states about 5X faster than the Pi4 but the miniPC is more than 2x faster than the Pi5. I haven't bought any new RPi lately but I am not impressed with their development. I look at specs when buying a device like a Pi and it seems that they have implemented the USB3 standard without any thought to getting the data off the port and into the system. They seem to be adding features at the expense of usability and overall functionality. While the Pi platform is still head and shoulders above the PC for GPIO work it is pushing itself out of the computer arena. No offense intended, AstroArch is a very well thought out and executed distro but the Pi platform comes up short supporting it.

Thank you all for your help sorting this out.

closed 2025-10-30 by MattBlack85