devDucks/astroarch #126: Problem with Update Astroarch

Olli781, 2024-12-22:
Hello everyone, when trying to update Astroarch the following error message appears:

From https://github.com/devDucks/astroarch
 * branch            main       -> FETCH_HEAD
Already up to date.
[sudo] password for astronaut:
:: Synchronizing package databases...
 astromatto.db failed to download
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
error: failed retrieving file 'astromatto.db' from astroarch.astromatto.com:9000                                                                                : Failed to connect to astroarch.astromatto.com port 9000 after 32 ms: Couldn't                                                                                connect to server
error: failed to synchronize all databases (unexpected error)
warning: archlinux-keyring-20241203-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) archlinux-keyring-20241203-1

Total Installed Size:  1.67 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n]
(1/1) checking keys in keyring                     [----------------------] 100%
(1/1) checking package integrity                   [----------------------] 100%
(1/1) loading package files                        [----------------------] 100%
(1/1) checking for file conflicts                  [----------------------] 100%
(1/1) checking available disk space                [----------------------] 100%
:: Processing package changes...
(1/1) reinstalling archlinux-keyring               [----------------------] 100%
==> Appending keys from archlinux.gpg...
==> Updating trust database...
gpg: next trustdb check due at 2025-01-01
:: Running post-transaction hooks...
(1/3) Reloading system manager configuration...
(2/3) Arming ConditionNeedsUpdate...
(3/3) Refreshing PackageKit...
:: Synchronizing package databases...
 astromatto.db failed to download
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
error: failed retrieving file 'astromatto.db' from astroarch.astromatto.com:9000 : Failed to connect to astroarch.astromatto.com port 9000 after 38 ms: Couldn't connect to server
error: failed to synchronize all databases (unexpected error)

The connection to the Internet works. Does anyone here have an idea?

MattBlack85, 2024-12-22:
Hi @Olli781, this has been brought to my attention yesterday. Shortly, we left for holidays and my son unplugged the pi where the package server is running 😛

I won't be back until the 30th so no updates will go through for now. Sorry for any inconvinience!

dividebysandwich, 2024-12-23:
Oh no! If only you had a pi cluster with duplicate PSU for redundancy...
...and two sons, for redundancy. ;)

supritsingh, 2024-12-27:
Hi Matt. I faced the same problem.. perhaps might be useful to have two package servers.. also could you let us know (through another readme file) how you maintain this server so that it can be replicated.

MattBlack85, 2024-12-27:
Hi @supritsingh ,the server is running on a pi within my home network. I will move the packages to a cloud bucket after my return and put a cname behind that so that it will be available 24/7/365 with no interruptions

MattBlack85, 2024-12-30:
all is up again, closing this

closed 2024-12-30 by MattBlack85