devDucks/astroarch #127: Can´t install Firecapture

miguegh, 2025-01-16:
Hi, could somebody install firecapture on astroarch?.  I have the astroarch installed and working on a Pi 4 8GB, Everything is fine but I´m trying to install firecapture package from AUR using paru -S firecapture and after  confirm the review it looks that nothing is happening.. I have tried and wait more than a hour and the process is still in the same aparent status.. 
Any idea?
Thanks!

MattBlack85, 2025-01-16:
@miguegh it should be available from my repo directly, can you check with pacman?

miguegh, 2025-01-17:
Hi @MattBlack85 oh... thanks... my mistake was try  with pacman without update.. received an error.. then I check the astroarch info and tried with paru.. 
Today I have updated astroarch and now with pacman has been correctly installed!!
Thanks!

MattBlack85, 2025-02-06:
happy it worked, closing this!

closed 2025-02-06 by MattBlack85