devDucks/astroarch #129: Updating and other issues.

vidrazor, 2025-01-28:
Hello, I've installed AstroArch on a Pi5, and I've been having a few issues with it.

First, I can't install programs from the terminal, and I can't install .deb files.

For instance, wen I type "sudo apt-get update", I get a message that says "sudo apt-get: command not found".  I get this also when trying to update the third party INDI drivers.

I tried to update AstroDMX, because I want to get away from that HIDEOUS blinding white interface (who thought that was a great idea?), but when I download and double click on the .deb file, the Discover Software Center program opens and an error window pops up saying "Could not open file, then lists the location of the download folder and and .deb package filename.

So I went back to the terminal and tried the "update-astroarch" command, which proceeded to run and extensive update. However when I rebooted the system, I got some distorted onscreen text, twice, and then, dead. As in, DEAD!

I had to reinstall the ISO from scratch, and start all over again, and I decided to just leave well alone this time around and not update. I just installed whatever packages I needed that were available to me on the Discover Software Center, such as FSearch, in addition to all the astro catalogs, and the astrometry files, and HIPS overlays that KStars needs that I have archived locally.

Finally, the system goes to sleep if there is no activity after x amount of time. Is there a way to shut this off? This is especially problematic if I don't connect to the unit in time via VNC viewer on a hardwired network. I either get a blank screen that I can't get out of, or I get a message telling me that the system is refusing to connect.

So, what gives? How do I update this system without blowing it up, and why isn't it recognizing standard commands like apt-get, or allowing me to install .deb files, and how do I turn off the sleep function?

Thanks.

LuckyEddie47, 2025-01-28:
Hi @vidrazor. From your writings you are familiar with the Debian parented Linux distributions - Debian, Ubuntu, Raspberry Pi, Mint, etc. AstroArch is (as the name suggests) is a distribution of Arch Linux. Arch uses a completely different system for packages and updates - neither .deb's nor .rpm's (the Red Hat family equivalent) will just work. Can I suggest that you have a quick look over the first few results on a search for "Arch Linux for beginners" or similar in your search engine of choice?

I'm sure that "others" will be along to answer your more specific frustrations such as the sleep options.

vidrazor, 2025-01-28:
Actually, I'm barely familiar with Linux. I only know of these commands because I've used them in the past with Astroberry. So all the command-line gobblety-gook is all alien to me. 

I happened to have another MicroSD card free, so I installed a fresh AstroArch from the ISO on it, and the first thing I did this time was to perform the "update-astroarch" command in the terminal, but after going through a bunch of different things that went on for some period of time, at the end of it all I got this message:

"error: failed retrieving file 'noto-fonts-1:2025.01.01-1-any.pkg.tar.xz' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed retrieving file 'archlinux-appstream-data-20241214-1-any.pkg.tar.xz' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed retrieving file 'mesa-1:24.3.4-1-aarch64.pkg.tar.xz' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
warning: too many errors from mirror.archlinuxarm.org, skipping for the remainder of this transaction
warning: failed to retrieve some files
error: failed to commit transaction (unexpected error)
Errors occurred, no packages were upgraded."

The last line says it all. I rebooted it and it miraculously came back up, the only thing different was a little box that popped up with AstroArch news. Out of curiosity, I went to the Discover Software Center, and was able to install the latest KStars, but it did not update the old installation, but made a second installation that has to be set up from scratch. I'm not sure if this also installed and is accessing a newer INDI database, or the same one. Unfortunately AstroDMX does not show up in the Discover Software Center.

So anyway, I'm not much ahead.

Thanks for your reply.

vidrazor, 2025-01-28:
Thanks for your reply.

1) Well, I simply don't know Linux, and thought those commands would work.

2) Well, seeing "Errors occurred, no packages were upgraded.", and seeing KStars had not been updated led me to believe something was wrong. That I was able to install the latest KStars from the Discover Software Center further led me to believe something was wrong. So hopefully in your update you'll have both KStars and the latest INDI driver base and yes, AstroDMX. :-) 

3) Honestly I don't know how long it takes to go to sleep. Once when I had it up and had the Pi 5 connected to a keyboard, mouse, and monitor, I came back into the room and the screen was black, not even a screen saver going. I moved the mouse and the system came back up. 

However when connecting via VNC through a local hardwired network, the first time it refused to connect to VNC, the second time I got a the blank screen, but mouse or keyboard action (via VNC) would not bring the system back. I had not immediately connected to the Pi because I was distracted by other things, and then experience this when I did try to connect. I have to reboot the Pi and connect as soon as possible, and then I'm able to connect via VNC. I have not yet let it go to sleep while already connected to VNC, so I don't know what the outcome of that will be. More importantly is when you're out in the field in a session, you don't want the system to go to sleep.

4) I have the entire astrometry database archived locally, as well as the entire HIPS All Sky Overlays, which are 32 and 21 gigs respectively, which I copy over to the system so KStars has all this data accessible when I am out in the field. This is not an issue.

vidrazor, 2025-01-29:
Hmm, well I ran "update-astroarch" on my test mule MicroSD card, and KStars updated, but AstroDMX is still the same version.

MattBlack85, 2025-01-29:
seems I compiled it but never updated the DB, it should be ok now, please try another update-astroarch

vidrazor, 2025-01-30:
OK, that worked, thanks. However, something weird is up, and I want your feedback on it.

I ran the update on the test mule MicroSD card and everything was great, no more blinding AstroDMX. However when I ran "update-astroarch" on the system I had already set up, I got the error(s) you see below. I decided to just sacrifice the system I spent days on getting ready and reflashed the card with the only available ISO I found, which was the same 1.9.0 on your download link. I ran "update-astroarch", and got the same error message.

So, question. Could a MicroSd card be a problem? It is a brand new PNY Elite-X card like this:
https://www.amazon.com/PNY-Elite-X-microSD-256GB-Class/dp/B079ZFFVYS?th=1

The test mule card is an older Sandisk Ultra, like this:
https://www.amazon.com/SanDisk-256GB-microSDXC-Memory-Adapter/dp/B0B7NV73PJ?th=1

I don't really understand those spec differences, so I don't know if the PNY is a lower spec card or not, but it appears to be the reason why I've had so many issues trying to get this system up.

Because everything seems fine with the Sandisk, I'm just gonna have the kid I've been setting this system up for buy my another Sandisk, I'll lick my wounds, and process to set up the system yet again. I can only imagine the problem has been this PNY card.

So thanks again for the update. Will a new ISO with the update be available soon so you don't need to run "update-astroarch" ?

BTW, I read that this ISO can also be used on an Intel machine? Wouldn't it need to be compiled for that hardware? I may load AstroArch into a MeLe Quieter computer I have, but I want to make sure the ISO will work with it.

Thanks again.

PR #163 (2025-07-18): Release 2.0
What's new:

- Fixed HDMI issue
- Fixed plasma x11 session issue
- Added visual tool to change settings
- Fixed xorg.conf config

Fix #159 
Fix #161
Fix #155
Fix #133
Fix #129
Fix #120