devDucks/astroarch #130: Catalog download issues

vidrazor, 2025-01-31:
OK, so I installed and updated AstroArch with the latest KStars 3.7.4 and the updated AstroDMX with the dark screen (thank you!), but now I'm having issues downloading the star catalogs under the DATA tab in KStars.

Certain catalogs, such as:
USNO NOMAD Catalog (Part)
Lynd's Catalogue of Dark Nebulae
ImagingPlanner Catalog1

...comeback with an "Install Again" prompt on the Install button for them. 

Other catalogs are listed as installed, but when I closed the "Download New Data" window, I got a series of successive boxes indicating:
Could not import "USNO NOMAD Catalog (Part)" Catalog. Catalog is not readable.
Could not import "Hicksopn Compact Groups" Catalog. Catalog is not readable.
Could not import "NGC IC (Streinicke)" Catalog. Catalog is not readable.
Could not import "Abell Planetary Nebulae" Catalog. Catalog is not readable.
Could not import "OpenNGC" Catalog. Catalog is not readable.

Other than the first three I list above with the "Install Again" prompt , they are all indicating that they are installed.

So are these catalogs installed or not? I had previously done this on the KStars 3.7.1 version before re-flashing and updating AstroArch on a different MicroSD card, and the catalogs all installed without issue. I can live without the "ImagingPlanner Catalog1", but the others should be in the system for KStars.

Thanks for your time.

vidrazor, 2025-02-06:
Assuming this may take a while to be resolved, I wiped out the updated system and reinstalled astroarch-1.9.0. with KStars 3.7.1. We'll simply have to deal with the blinding AstroDMX screen LOL. 

More importantly however, all the the catalogs readily load into KStars 3.7.1 without issue. I've also installed the entire astrometry and HIPS databases and will just proceed to use this system.

sc74, 2025-02-06:
I just installed the catalogs on 3.7.5 without problems. Could be a server problem when downloading

vidrazor, 2025-02-07:
3.7.5? Anyway, 3.7.1 actually works fine, our real objection is the white AstroDMX screen which comes loaded with the astroarch-1.9.0 ISO. We'll just have to deal with it, unless there's a way to just update AstroDMX without having to go through an entire system update. Everything else works just fine.

MattBlack85, 2025-02-07:
@vidrazor to update a single package you can run sudo pacman -Sy (check and sync the DB with the latest versions of all packages) and then sudo pacman -S PKG_NAME for a single package

vidrazor, 2025-02-08:
I tried to update the single package, but it kept telling me there was no such package.

However realizing there is now KStars 3.7.5, I took a change (again, LOL) and ran "update-astroarch", and this time the update went without a hitch. 

Version 3.7.5 did not exhibit the download issue with the catalogs, so now all is well all around.

Thanks for your time.

closed 2025-02-08 by vidrazor

MattBlack85, 2025-02-08:
happy all is good now