devDucks/astroarch #131: Going around in circles :-)

vidrazor, 2025-02-09:
Man, I feel like Ground Hog's Day with AstroArch on a Pi 5.

I had everything up, working beautifully, then I turn the unit on and...nothing. I can't connect to it, either through a hardwired network (which I have been doing all along), and there's also no WiFi coming from it. Dead.

So I decide to flash an entirely different MicroSD card with ISO 1.9, then run "update-astroarch", it goes through it's motions, and I wind up with what you see below at the end.

So is this a Pi 5 thing? Will this ISO load into an Intel MiniPC instead? Why do these things keep on happening?

Thanks for your time.

Total Download Size:   1395.82 MiB
Total Installed Size:  5755.13 MiB
Net Upgrade Size:       594.76 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 kstars-3.7.5-1-aarch64                111.2 MiB  2.91 MiB/s 00:38 [------------------------------------] 100%
 chromium-131.0.6778.264-1-aarch64      85.7 MiB  1338 KiB/s 01:06 [------------------------------------] 100%
 astro_dmx-2.12.1-1-aarch64             45.4 MiB  4.62 MiB/s 00:10 [------------------------------------] 100%
 breeze-6.2.5-1-aarch64                 88.4 MiB   967 KiB/s 01:34 [------------------------------------] 100%
 qt6-webengine-6.8.2-1-aarch64          72.3 MiB   759 KiB/s 01:38 [------------------------------------] 100%
 firefox-134.0.2-1-aarch64              56.1 MiB   973 KiB/s 00:59 [------------------------------------] 100%
 linux-firmware-20250109.7673dffd...   163.7 MiB  1558 KiB/s 01:48 [------------------------------------] 100%
 llvm-libs-19.1.7-1-aarch64.pkg.tar.xz failed to download
 Total (  8/693)                      1395.8 MiB  13.0 MiB/s 01:48 [------------------------------------] 100%
error: failed retrieving file 'llvm-libs-19.1.7-1-aarch64.pkg.tar.xz' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
warning: failed to retrieve some files
error: failed to commit transaction (unexpected error)
Errors occurred, no packages were upgraded.
----------------------------------------------------

sc74, 2025-02-09:
The download stops due to a timeout when loading for too long. 
In a console, open the pacman configuration file with the command : sudo nano /etc/pacman.conf
Then add the value below, just after the line "ParallelDownloads = 5"

DisableDownloadTimeout 

Save the file with the ctrl+x+enter command

Restart your update-astroarch

vidrazor, 2025-02-09:
Thanks. It appears perhaps an issue at the server end? I tried the update today and the updates went through, now the new card is fully updated.

However the system blacking out is a concern. Everything was working fine with that setup, had everything loaded and hardware set up, and then everything went dark. No network, no WiFi. I noticed when the system was starting up, the drive light would shut off for a few moments, then come back on, and that was it.  I'll see how it goes with this new card.

Thanks again for your time.

sc74, 2025-02-10:
Maybe the pi documentation on leds could help you if you have a breakdown https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes

Jimbob-B, 2025-02-22:
I think this may be the broken `mirror.archlinuxarm.org` issue where it doesn't quite work properly to point you to the nearest mirror. I would manually configure the mirrors by reading this,

[Mirrors - ArchWiki](
https://wiki.archlinux.org/title/Mirrors)

closed 2025-09-03 by MattBlack85