devDucks/astroarch #132: RTC not recognized

vidrazor, 2025-02-15:
I've been wanting to add a clock backup for a Pi5 and I purchase a DS3231 RTC Module for it. I plugged into inner pins 13579 on the Pi, and went through the instruction here on how to set it up. So I booted and typed "sudo i2cdetect -y 1", and got the same number 68 as in your example.

However, when I went to add the ds3231 number to the dtoverlay=i2c-rtc in the config.txt file, ds3231 was already there. 

So according to the instructions, I should be seeing UU when I type in sudo i2cdetect -y 1, because ds3231 was already there. All I get however is the number 68.

So is the RTC board I got not compatible with the Pi 5? The board I got is this one:

https://www.amazon.com/dp/B08X4H3NBR?th=1

Thanks for your time.

vidrazor, 2025-02-19:
So do I take it that you can't use a DT3231 board on a Pi 5?

Jimbob-B, 2025-02-22:
I'm not sure if I'll be much help as I've not tried running astroarch on my pi5 but couple of things I can think of when reading your question are,

- The Pi5 has a built in RTC and I think you will find this will be assigned `/dev/rtc0` and your DS3231, when recognized, will be `/dev/rtc1`. What is listed under `/dev/rtc*` now? The linux kernel is usually coded to pickup `rtc0` as the default and, from what I can see when messing about with a Tinkerboard, you can't easily change it to use `rtc1` unless you recompile the whole kernel and change the default RTC setting.
- That module should work fine ... I have a similar one running on my StellarMate install that I purchased from thepihut.com. This is a Pi4 that I dual boot between StellarMate and AstroArch. I would check what the kernel is recognising for `rtc` devices by using `dmesg | grep -i rtc`. You *should* see entries for both the Pi5 RTC and the DS3231 module. The kernel messages may help debug what is happening to cause the issues you're seeing. The Pi5 `rtc0` module will look something like this,

```
[    0.258946] rpi-rtc soc:rpi_rtc: trickle charging enabled at 3000000uV
[    0.264144] rpi-rtc soc:rpi_rtc: registered as rtc0
[    0.265527] rpi-rtc soc:rpi_rtc: setting system clock to 2025-02-22T15:56:05 UTC (1740239765)
```
This is assuming the Arch Linux kernel recognizes the Pi5 RTC.