devDucks/astroarch #133: Problem with the latest update

rojofree, 2025-02-19:
Hello. With the latest update, I have no video output. It just displays what is seen on the right monitor. Instead, it works correctly with VNC (notebook on the left).
Do you know a solution?
I deleted and installed everything from scratch. Losing all the configuration.
It happened again.
It works perfectly until the last batch of updates that left it like this.

Thanks

Pi 5 - 8gb
NVME disc
Case Neo

[image]
[image]

[image]

vidrazor, 2025-02-19:
Strange that you can still connect via VNC. I got that once, and I had jack, it was just dead in the water. Had to reinstall from scratch.

rojofree, 2025-02-19:
> Strange that you can still connect via VNC. I got that once, and I had jack, it was just dead in the water. Had to reinstall from scratch.

Hi, I've tried that. It's the only thing I tried. It worked fine after reinstalling. But with the latest update (after rebooting) the problem came back. It must be a resolution issue with the signal being output to the external monitor.

rojofree, 2025-02-19:
I tried another monitor with a different resolution. With another HDMI cable. Same error.

sc74, 2025-02-19:
The latest update of the kernel and video drivers breaks the HDMI configuration. You need to modify several files to find a good config or wait for the new version of AstroArch.

These are the following 4 files:

/boot/config.txt
/etc/X11/xorg.conf.d/99-v3d.conf
/etc/X11/xorg.conf
/home/astronaut/.config/kwinrc

You can find the updated versions here

https://github.com/devDucks/astroarch/tree/1.10/configs

rojofree, 2025-02-21:
I'll wait, since I don't know how to make the changes.

Thanks

PR #163 (2025-07-18): Release 2.0
What's new:

- Fixed HDMI issue
- Fixed plasma x11 session issue
- Added visual tool to change settings
- Fixed xorg.conf config

Fix #159 
Fix #161
Fix #155
Fix #133
Fix #129
Fix #120