devDucks/astroarch #135: Possibility to update Indi drivers

HLex79, 2025-03-08:
Hi,

thanks for this very stable version to run all our beloved tools.

I am shiny new to all of this so please do not make jokes :-)

Is there a way to update the indi libs and drivers seperately? How does this work?

As an example, I want to update just the touptek drivers and the onstep OCS? Does this work?

When I run the update command having selected the bleeding edge, does this pull the latest and greatest or how does this work?

BR and thanks!

Lex

sc74, 2025-03-14:
See this link to update the indi drivers individually.

https://github.com/indilib/indi-3rdparty#building-individual-3rd-party-drivers-typically-recommended

I don't recommend this, as it's not really suitable for a novice.

If you want to use the development versions of indi/kstars, use the use-astro-bleeding-edge command in a terminal. If that doesn't work, you can revert to the stable versions with use-astro-stable.

See this link:
https://github.com/devDucks/astroarch?tab=readme-ov-file#useful-commands

MattBlack85, 2025-04-25:
@HLex79 this is not possible with astroarch at the moment and I am not sure it it will ever be available. The problem is not a huge one but is tedious and time consuming to rework the pipeline to compile every single package as a standalone one. I may be keen to go that way if someone would be willing to mantain a given subset of the drivers build files and if we can make it to a point where building those thing is very automatic (it's all manual at the moment)