devDucks/astroarch #136: Update is not performed

HLex79, 2025-03-10:
Hi,

I do not know if it is only me but up on the last notification i tried the update-astroarch command.
The process shows that apparently everything is already up to date.
After checking KSTARS the version is not on 3.7.5?

Am I doing something wrong?

BTW, thanks again for this wonderful solution!

BR,

Lex

sc74, 2025-03-14:
When you run update-astroarch, do you see the following lines?

:: Synchronizing package databases...
 astromatto is up to date

If so, did you run the update-astroarch command again?

HLex79, 2025-03-14:
Yes I see the above mentioned lines. Tried it at least 5 times. Restarted my RPi5 a couple  after trying the update. Sadly KSTARS stays at version 3.7.2 Beta?

sc74, 2025-03-14:
Maybe you could try the astro-rollback-full command and then do an astroarch-update again

HLex79, 2025-03-15:
If I perform astro-rollback-full I get this in console:

loading packages...
resolving dependencies...
looking for conflicting packages...
:: indi-3rdparty-drivers-2.1.1-1 and indi-3rdparty-drivers-git-r122.3e5c7b3-1 are in conflict. Remove indi-3rdparty-drivers-git? [y/N]
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: indi-3rdparty-drivers-2.1.1-1 and indi-3rdparty-drivers-git-r122.3e5c7b3-1 are in conflict
--2025-03-15 10:25:48--  http://astromatto.com:9000/aarch64/kstars-3.7.4-2-aarch64.pkg.tar.xz
Resolving astromatto.com (astromatto.com)... 91.218.202.244
Connecting to astromatto.com (astromatto.com)|91.218.202.244|:9000... connected.
HTTP request sent, awaiting response... 200 OK
Length: 116676060 (111M) [application/octet-stream]
Saving to: ‘kstars-3.7.4-2-aarch64.pkg.tar.xz’

kstars-3.7.4-2-aarch64.pkg.tar. 100%[====================================================>] 111.27M  9.66MB/s    in 12s

2025-03-15 10:26:00 (9.50 MB/s) - ‘kstars-3.7.4-2-aarch64.pkg.tar.xz’ saved [116676060/116676060]

loading packages...
resolving dependencies...
looking for conflicting packages...
:: kstars-3.7.4-2 and kstars-git-r122.3e5c7b3-1 are in conflict. Remove kstars-git? [y/N]
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: kstars-3.7.4-2 and kstars-git-r122.3e5c7b3-1 are in conflict

I previously reinstalled with balena because i though i messed something up, while upgrading drivers seperately. The message above was displayed also before my reinstall...

BR,

Lex

HLex79, 2025-03-15:
After a couple of back and forth, running roll back and update commands and rebooting between, the errors above are now gone. 
But sadly no new kstars version, still on 3.7.2 beta...

sc74, 2025-03-15:
I don't understand. You flashed your SD card with Astroarch 1.9 and just applied the update-astroarch command? Or did you do other commands? Could you make screenshots?

HLex79, 2025-03-15:
Yes, I flashed my nvme (with the steps mentioned in the guide), selected bleeding, tried to update..

[image]

[image]

HLex79, 2025-03-15:
[image]

sc74, 2025-03-15:
Okay, I see. The beta version of kstars for arm hasn't been updated on the astromatto repo. If you want the latest version of kstars, you have to stay on the stable version. Run the use-astro-stable command to return to the initial state of the installation

HLex79, 2025-03-15:
Thank you!
In the end Kstars it self is not so important for me (even if i love new SW :-), the drivers of 3rd party are more.
When I check a driver version (OCS that i really need to start building my ROR Obsy) in ekos, i see it is not updated.
What do you suggest?

HLex79, 2025-03-15:
When I try to astro-stable, i get this:

[image]

Really sorry to bother you...

sc74, 2025-03-15:
OCS is part of indi, which is up-to-date with 2.1.2. When you open the indi control panel, it's not up-to-date?
For OCS, you should post your questions on the forum https://indilib.org/forum
I think they can help you with your ROR Obsy

Delete the file with the command sudo rm /usr/lib/libqhyccd.so.20 then redo use-astro-stable

HLex79, 2025-03-15:
You rock! 
I am now on stable KStars 3.7.5 and I updated OCS separately.
In theory, green light to start the ROR buidl!

Just to be sure, I should now never return to bleeding edge?

Thank you very much for your help, appreciate!

sc74, 2025-03-15:
It's perfect if everything works. Keep the stable version for your observatory project. The bleeding version is more for experts to do tests

MattBlack85, 2025-04-25:
closing as the client is happy 🥳

closed 2025-04-25 by MattBlack85