devDucks/astroarch #137: System update issue

Astrostecco, 2025-03-20:
Hi,

I tried updating to the latest release, but the process stops after several packages apparently cannot be downloaded (see screenshot).
Is this a problem with some un-reachable server, and I should only wait? Or is it a problem with my specific machine?

Thanks!
Stefano

[image]

Astrostecco, 2025-03-20:
Nevermind my post....I re-run the update command and this time went through with it

[image]

I did re-run the command already though, and every time was stopping at a different download point....it seems a bit random, not sure what is causing it?

Best,
Stefano

imahgin, 2025-03-20:
Yes I faced the same issue, and eventually, after a few attempts, everything was updated successfully.
It often happens to me and I solved it by trying several times

sc74, 2025-03-20:
This issue is known. Simply add the following line to the /etc/pacman.conf file after ParallelDownloads=5:

DisableDownloadTimeout

Astrostecco, 2025-03-22:
Thank you Mattia!

MattBlack85, 2025-04-25:
it's all thanks to @sc74 :)

closed 2025-04-25 by MattBlack85