devDucks/astroarch #138: Conflicting dependencies

dedo70, 2025-03-23:
Hello Mattia,

I have some conflicting dependencies which prevents Astroarch from being upgraded.

These are the messages I receive when issuing update-astroarch:

:: Vuoi sostituire tracker3 con extra/tinysparql? [S/n] y
risoluzione delle dipendenze in corso...
ricerca dei pacchetti in conflitto in corso...
:: tinysparql-3.8.2-3 e tracker3-3.7.3-2 vanno in conflitto. Rimuovere tracker3? [s/N] y
errore: sono stati rilevati dei conflitti irrisolvibili tra i pacchetti
errore: impossibile eseguire l'operazione richiesta (dipendenze in conflitto)
:: tinysparql-3.8.2-3 e tracker3-3.7.3-2 vanno in conflitto (tracker3<=3.7.3-2)

Thank you so much for all your effort!
Andrea

sc74, 2025-03-24:
If you've changed the system language according to the ArchLinux wiki, you need to delete a KDE Plasma file to resolve this error.

rm ~/.config/plasma-localerc

dedo70, 2025-03-24:
It worked! Thank you Stéphane

closed 2025-03-24 by dedo70