devDucks/astroarch #140: No mount or camera in Ekos

Trout397, 2025-03-31:
Hi, I've had a problem where all seems to be well until I connect equipment. 

Ekos opens up and I have entered a profile with equipment specified. I go to connect and external equipment does not connect, gpsd does as well as astrometry. However my mount and camera don't connect. If I go through manually connecting them they do connect but no mount or camera tabs are shown in Ekos.

Any ideas?

Cheers 

Steve

sc74, 2025-04-02:
Could you give us more information? Camera type, mount? Screenshots?

Trout397, 2025-04-02:
Hi, the camera is Altair astro 294c and the mount is EqMod.
Sorry I can't do a screen shot as I'm away from my setup at the moment.
Thanks,
Steve

Trout397, 2025-04-16:
Hi,

I've now a screenshot. My mount and ann Altair 183c are connected but don't show up in the Ekos pane. Any ideas.

Steve

[image]

sc74, 2025-04-16:
Why are you trying to connect to a QHY camera? Did you choose the correct Altair driver for your camera? Could you show your profile window?

Trout397, 2025-04-16:
Hi, I think I might have worked it out.

When Ekos is unable to connect everything in the profile the other tabs
don't open. Mount etc. When everything connects all of the tabs open up.
This didn't happen in the version of Ekos on astroberry, anything that
connected got it's tab shown.

The qhy is the guide camera, by the way.

Hopefully this is sorted out now,
I'll give it a try when the skies are clear again.

Many thanks for your time,

Steve

MattBlack85, 2025-04-25:
closing as non-bug, feel free to re-open if the problem persists

closed 2025-04-25 by MattBlack85