devDucks/astroarch #141: Chromium won't start.

Jimbolaya64, 2025-04-01:
Error is:

~ » chromium
/usr/lib/chromium/chromium: error while loading shared libraries: libicui18n.so.75: cannot open shared object file: No such file or directory

Using astroarch 1.9.0. Please advise, thank you.

sc74, 2025-04-02:
This is not an AstroArch OS issue. The chromium package for arm has not been updated by the maintainer yet. We have to wait because this package upgrade work is done on the maintainers' free time and for free for the ArchLinux community. You can look here https://archlinuxarm.org/packages/aarch64/chromium to see the version which is at 131 and wait for it to go to 134

Jimbolaya64, 2025-04-02:
Got it! Thank you.

closed 2025-04-25 by MattBlack85