devDucks/astroarch #142: v1.9.3 and libcamera issues (imx477) again

ArnaudMarot, 2025-04-04:
Good day all,
I've a problem with libcamera / indi_libcamera strange behavior.
Fresh install of astroarch and updated to 1.9.3.
Tried to configure the IMX477 CSI cam (Innomaker Raspi HQ camera clone.)
->update /boot/config.txt with 
camera_auto_detect=0
add dtoverlay=imx477

With the up to date astroarch distro, I've the libcamera 0.4.1 installed
using cam -l find the camera and is working.

~ » cam -l                                                                                                                                                                                                              astronaut@astroarch
[0:12:52.977443798] [2163] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:12:52.977545685] [2163] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:12:52.977609944] [2163]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0
[0:12:53.014873670] [2164]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:12:53.017147508] [2164]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media2
[0:12:53.017376338] [2164] ERROR Pipeline pipeline_handler.cpp:629 Configuration file '/usr/share/libcamera/pipeline/virtual/virtual.yaml' not found for pipeline handler 'virtual'
[0:12:53.017400226] [2164] ERROR Virtual virtual.cpp:336 Failed to open config file:
Available cameras:
1: 'imx477' (/base/soc/i2c0mux/i2c@1/imx477@1a)

Installed rpicam-apps 1.5.1-1
But using rpicamera-hello --list-cameras gives me errors and no image received.
also gives me error of dma_heap not available.

rpicam-hello --list-cameras
Could not open any dmaHeap device
rpicam-hello: /usr/include/libcamera/libcamera/controls.h:166: T libcamera::ControlValue::get() const [with T = std::__cxx11::basic_string<char>; typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `type_ == details::control_type<std::remove_cv_t<T>>::value' failed.
[1]    1573 IOT instruction (core dumped)  rpicam-hello --list-cameras

tried everything i could find
modifying cmdline.txt
adding cma=256M (tried 256/400/512 without improvement)
adding a picamera-rules
I've seen the issue #104 with the fix (also tried it as there is no picamera.rules in udev with the fresh install. (But the problem I have is slightly different: even logged as root, the dma_heap problem remains.)
libcamera still complaining that no dma_heap device available (whatever the user astro/root)
I've also discovered that indi_3rdparty_drivers are not complete. (ver 2.1.2-1 installed)
in the fresh install, indi_libcamera driver is missing.
Had compiled indi_libcamera and installed (struggling with dependencies, but finally compiled without errors and installed)
But never worked properly, it seems that, despite the fact the camera is recognised, the driver is pointing to a virtual device and doesn't find the proper configuration file.
and still, the dma_heap problem remains.

~ » indiserver -vvv indi_libcamera_ccd                                                                                                                                                                                  astronaut@astroarch
2025-03-07T01:29:17: startup: indiserver -vvv indi_libcamera_ccd
2025-03-07T01:29:17: Driver indi_libcamera_ccd: pid=2247 rfd=6 wfd=6 efd=7
2025-03-07T01:29:17: listening to port 7624 on fd 5
2025-03-07T01:29:17: Local server: listening on local domain at: @/tmp/indiserver
2025-03-07T01:29:17: Driver indi_libcamera_ccd: sending msg nq 1:
<getProperties version="1.7"/>

2025-03-07T01:29:18: Driver indi_libcamera_ccd: Could not open any dmaHeap device
2025-03-07T01:29:18: Driver indi_libcamera_ccd: [0:17:16.829490252] [2247] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
2025-03-07T01:29:18: Driver indi_libcamera_ccd: [0:17:16.829560325] [2247] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
2025-03-07T01:29:18: Driver indi_libcamera_ccd: [0:17:16.829638213] [2247]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0
2025-03-07T01:29:18: Driver indi_libcamera_ccd: [0:17:16.941858849] [2248]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
2025-03-07T01:29:18: Driver indi_libcamera_ccd: [0:17:16.947080089] [2248]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media2
2025-03-07T01:29:18: Driver indi_libcamera_ccd: [0:17:16.947416880] [2248] ERROR Pipeline pipeline_handler.cpp:629 Configuration file '/usr/share/libcamera/pipeline/virtual/virtual.yaml' not found for pipeline handler 'virtual'
2025-03-07T01:29:18: Driver indi_libcamera_ccd: [0:17:16.947461435] [2248] ERROR Virtual virtual.cpp:336 Failed to open config file:
2025-03-07T01:29:18: Driver indi_libcamera_ccd: indi_libcamera_ccd: /usr/include/libcamera/libcamera/controls.h:166: T libcamera::ControlValue::get() const [with T = std::__cxx11::basic_string<char>; typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `type_ == details::control_type<std::remove_cv_t<T>>::value' failed.
2025-03-07T01:29:19: Driver indi_libcamera_ccd: read: Connection reset by peer
2025-03-07T01:29:19: Driver indi_libcamera_ccd: restart #0
2025-03-07T01:29:19: Driver indi_libcamera_ccd: pid=2261 rfd=7 wfd=7 efd=9
2025-03-07T01:29:19: Driver indi_libcamera_ccd: sending msg nq 1:
<getProperties version="1.7"/>

Also tried to install indi-pylibcamera as stated in astroarch readme, but it breaks the distro and have to rollback libcamera0.3.0
I also wonders if the libcamera 0.4.0-1 installed is not the rpi branch, but the 'standard' one? (IMHO the libcamera jungle is quite messy...)
and moreover has some dependencies problems (libboost_program_options 1.86.0 needed, installed 1.87.0).
and by adding symbolic links to 1.86, it's still not working, despite the fact that libcamera is now giving images.
With the rollback to libcamera 0.3, I've succeeded to receive camera images with libcamera tools
I've also succeeded once to use indi_libcamera driver with astrodmx. Successfully received images using indi_libcamera drivers, and then restarted the pi: back to the mess, nothing is working anymore. Dma_heap mess, astrodmx complaining that no buffer available. and no more images received. Despite the fact that, with libcamera tools, I'm still able to receive images from camera (despite the dma_heap complain... )
Very strange behavior,
I tried so much things, without success, that I don't know anymore what to try.

I've actually 2 differents installs of astroarch for trials
install A: 1.9.3 with libcamera0.3 /indi_pylibcamera installed -> the only small "success" i've had receiving images from the camera
install B: 1.9.3 with libcamera0.4 (no rollback) / only indi_libcamera installed, no indi_pylibcamera

the two installs have exact same configuration except for the libcamera version and related tools

Any help is welcome as this point. I'm fairly new to indi system, and I'm struggling with the docs and infos I can find on the internet regarding raspberrypi and imx477 /HQ cam issues, but no luck til now. Unfortunately, most of the informations found is related to *buntu/debian clones, and quite useless for arch distro.
I dream of a distro-agnostic doc for libcamera ecosystem...

I even thought to completely wipe out arch installs and install other system such as astroberry or similar, but I'd prefer to stick with archlinux as it's the ditro I already use on my other computers. (And I really like arch ;)

thanks
arno

sc74, 2025-04-04:
Maybe you could test the few packages I compiled here https://github.com/sc74/sc74.github.io/tree/main/aarch64
You install the packages with pacman -U package name

ArnaudMarot, 2025-04-04:
Hi, thanks for the answer.
Started to try your packages. Libpisp & rpicam-apps-> there is an improvement, I don't have the libcamera/controls.h errors anymore, but am still stuck with the dmaheap buffer 

~/builds/sc74.github.io/aarch64 (main*) » rpicam-hello --qt-preview                             astronaut@astroarch
Could not open any dmaHeap device
[0:48:10.529177224] [3720] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:48:10.529227020] [3720] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:48:10.529263871] [3720]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0
[0:48:10.573548404] [3722]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:48:10.575948018] [3722]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media2
[0:48:10.576137923] [3722]  INFO RPI pipeline_base.cpp:1118 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:48:10.576456826] [3722] ERROR Pipeline pipeline_handler.cpp:629 Configuration file '/usr/share/libcamera/pipeline/virtual/virtual.yaml' not found for pipeline handler 'virtual'
[0:48:10.576511029] [3722] ERROR Virtual virtual.cpp:336 Failed to open config file: 
Made QT preview window
Mode selection for 2028:1520:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 3456.22
    SRGGB12_CSI2P,2028x1080/0 - Score: 1083.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 0
    SRGGB12_CSI2P,4056x3040/0 - Score: 887
[0:48:10.831930079] [3720]  INFO Camera camera.cpp:1202 configuring streams: (0) 2028x1520-YUV420 (1) 2028x1520-SBGGR12_CSI2P
[0:48:10.832624903] [3722]  INFO RPI vc4.cpp:570 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
dmaHeap allocation failure for rpicam-apps0
ERROR: *** failed to allocate capture buffers for stream ***

It drives me crazy...

sc74, 2025-04-04:
Have you installed the libcamera-rpi, libcamera-ipa-rpi and libcamera-tools-rpi package? And dtoverlay=imx477 in the config.txt file?

ArnaudMarot, 2025-04-04:
Not yet as I tried your files with my install already containing libcamera 0.4 version. (and pacman is complaining about the missing libcamera 0.3 dependencies when I try to install your libcamera-rpi version.
But will try with the other install containing libcamera0.3
regarding dtoverlay, yes, it's in my config.txt

sc74, 2025-04-04:
Caution: You need libcamera-rpi, not libcamera. Uninstall all libcamera packages before installing libcamera-rpi packages

ArnaudMarot, 2025-04-04:
Okay, I did again remove libcamera-rpi & libcamera-ipa-rpi, and reinstalled your version: and i works! don't know what I messed up at first try. Now, It's a success! re-compiled the indi_libcamera driver, and it's perfectly working with kstars/ekos!
Now, I'll try to do the same with the other install (which has libcamera-0.4 installed)
But it seems astroarch install is a little bit messy: if i remember well, I didn't install libcamera manually, but it has been installed with update-astroarch: and the version installed is not libcamera-rpi. and btw, the indi-3rdparty-drivers is not complete (missing indi-libcamera, at least). Don't know why it's removed?
anyhow, thanks a lot for your help.

ArnaudMarot, 2025-04-04:
and additionally, I did remove the cma=256M in cmdline.txt : wasn't necessary.

sc74, 2025-04-04:
I don't remember exactly. Maybe it's from a discussion on the AstroArch discord https://discord.gg/JXsrMP9Y or issue #104. It's perfect if it works for you

ArnaudMarot, 2025-09-20:
HI, sorry not answering sooner,  I wasn't home and didn't had access to my astroarch install for a while.
Back home now, will give it a try.
thanks