devDucks/astroarch #143: Problem with serial connection on Pi5

miguegh, 2025-04-16:
Hi All, 
Recently I have purchased a Pi5 8GB version. I have been working with a Pi4 8GB good and without problems but with the Pi5 I´m having some issues, the most important one is related with the serial connection. This is the equipment that I´m trying to manage:
Mount: Celestron AVX with a Nexstar HC with RJ port (the classic) controlled via a Prolific USB to Serial adapter specific for Celestron (works perfect on Astroarch in the Pi4)
Camera: Player One Saturn-C SQR
Focuser: Celestron Focus motor for Schmid-Cassegrain (SCT)

The main problem is on the connection to the mount and focuser. But the origin looks to be on the serial ports. When I plug the "Prolific USB to Serial converter for Celestron Nexstar", this is the unique port that the system finds.  And when I try the first connection Kstars launchs the port selector offering the unique port to connect the mount (Nexstar HC) and the focuser (SCT).. and of course only is possible to connect with one. 

[image]

If I unplug the Prolific USB to Serial converter I have another issue: NO CANDIDATE PORTS to connect the SCT focuser and also no possibility to connect the mount. 

[image]

The camera is connecting perfectly and running on USB 3.0 port. 
I´m lost... it looks that the USB ports are not running correctly on the Pi5..???  . One more time... **no issue on Pi 4**

Another issue detected during the update-astroarch attempt: I needed to execute the command several times to complet the full update.. many errors during the process.. I don´t know why but after executing the command several times I got a full update of the system. So system is up-to-date. 

I will appreciate your comments.. I will still fighting with this..
Miguel

imahgin, 2025-04-16:
Hola tocayo,

The update issue is documented and easy to solve.
Just add the following line to the /etc/pacman.conf file after ParallelDownloads=5:
       DisableDownloadTimeout

Related to the serial USB port, I own a RBP5 and have no issues with the serial port on my EQ Mount.
However, I'm using one serial port. 
Have you tried to connect only the mount? I guess it works fine, isn't it?
What is the result of **_lsusb_** with both devices connected?

BR
Miguel

miguegh, 2025-04-16:
I have checked a Shoestring FCUSB and works without problem.. so I can not understand what could be causing the problem exactly between the Pi5 and the Celestron´s SCT focuser on USB..  which works on Pi4

miguegh, 2025-04-16:
> Hola tocayo,
>
> The update issue is documented and easy to solve. Just add the following line to the /etc/pacman.conf file after ParallelDownloads=5: DisableDownloadTimeout
>
> Related to the serial USB [...]

Hi Miguel, thanks for your reply, 

The update issue: yes.. I have noted that...I did not read correctly before comment this.. thanks.

I can connect with the mount without problem using the Prolific USB to Serial adapter... but only with the mount, not with the mount and the focuser, because both try to connect through the same port. This port is the one generated by the Prolific USB to serial adapter. And when I ask EKOS to scan for ports trying to get a port for the SCT focuser "no candidates"... 
I will check lsusb .. thanks!
Miguel

sc74, 2025-04-16:
This might give you an idea of ​​your issue https://www.cloudynights.com/topic/948924-celestron-sct-focuser-incompatible-with-raspberrypi5-usb/

miguegh, 2025-04-21:
> This might give you an idea of ​​your issue https://www.cloudynights.com/topic/948924-celestron-sct-focuser-incompatible-with-raspberrypi5-usb/

Hi @sc74 and thank you very much to share this topic.. finally I was in the correct line: no compatible with Pi5.. 
Ok, noted.. thank you very much.. to All.. 
Miguel

closed 2025-04-25 by MattBlack85