devDucks/astroarch #144: Installing packages, ftp, and updates

pippooo26, 2025-04-22:
Hi everyone,

I installed astroarch on RBPi 4B and am loving it. I am having issues installing packages. I tried using sudo pacman -S <package name> but it never manages to download the package. This also happens with the default Astroarch commands. For instance, ftp_on fails because when installing the vsftpd package it doesn't manage to download (gives connection timeout). The same happens when I try to update astroarch. This time what happened is that INDI was uninstalled and then it couldn't reinstall it. So I ended up having to re-flash astroarch.

I already tried editing the pacman.conf file with DisableDownloadTimeout and didn't help. All I need to do is to activate ftp and install PHD2 logviewer. Any suggestions (preferred if it doesn't involve astroarch update, since everything works great)?

Thanks!

sc74, 2025-04-22:
Did you run an update-astroarch after installing the image? Could you take screenshots of your issue?

pippooo26, 2025-04-22:
I did and what happened is that it uninstalled indi and couldn't install the new version. After trying several fixes, I just had to re-flash astroarch. Also, I had a lot of issue with INDI and my specific equipment in the past. The non-updated astroarch version works great, so if I could avoid updating that would be best.

I don't have my PI with me right now, sorry, but I will upload a screenshot tonight.

Thanks!

pippooo26, 2025-04-24:
Sorry for the delay. A screenshot of my issue when trying to turn ftp on is below. When I tried to update, the error I got was very similar (I'm not trying again to avoid having to flash everything again and lose my INDI settings).

[image]

pippooo26, 2025-04-24:
Screenshots for the result of sudo pacman -Fy and the ping are attached. I also attach a text file with the entire content of the pacman.conf file.

[pacman.conf.txt](https://github.com/user-attachments/files/19897360/pacman.conf.txt)

[image]
[image]

sc74, 2025-04-24:
Are you on a fresh install? Could you provide the output of systemctl status systemd-resolved?

pippooo26, 2025-04-24:
Yes I'm on a fresh install, not updated, nothing installed on top. I just used Kstars and PHD2 a couple of times after installing and that's it. Here's the output:

[image]

sc74, 2025-04-24:
Try sudo pacman -S archlinux-keyring --noconfirm then do sudo pacman -Fy again

pippooo26, 2025-04-24:
Same issue:

[image]

sc74, 2025-04-24:
And what did sudo pacman -S archlinux-keyring give?

sc74, 2025-04-24:
What does systemctl status dhcpcd give?

pippooo26, 2025-04-24:
This:
[image]

sc74, 2025-04-24:
Could you give the output of cat /etc/resolv.conf?

pippooo26, 2025-04-24:
[image]

sc74, 2025-04-24:
Your file is out of date. Could you try sudo systemctl restart systemd-resolved ?

pippooo26, 2025-04-24:
Here's the output of cat /etc/resolv.conf after running  systemd-resolved:

[image]

sc74, 2025-04-24:
Have you tried sudo pacman -Fy again?

pippooo26, 2025-04-24:
yes, exact same output

sc74, 2025-04-24:
Try adding nameserver 8.8.8.8 to the /etc/resolv.conf file. And test again

pippooo26, 2025-04-24:
This partially solved the problem! I still get an error message about a server not found:

[image]

pippooo26, 2025-04-24:
I was able to turn ftp_on! Thank you so much for your help! Can I ask you to just explain what the problem was? I'm trying to get more familiar with Linux, but it's a bit of a steep learning curve...

pippooo26, 2025-04-24:
(I tried to install cmake but I still have errors there:)

[image]

sc74, 2025-04-24:
I think that since you are not updating the whole system with update-astroarch, the resolv.conf file is not correct. This prevents domain name resolution and you are stuck on your local IP. For package download, the 404 error comes from the fact that the pacman database is not up to date. You need an up-to-date system to install packages like cmake

pippooo26, 2025-04-24:
I see, that makes sense, thanks! For the previous error (which we fixed by adding 8.8.8.8 to the resolve.conf), what do you think caused that? That was preventing me from updating astroarch as well...