devDucks/astroarch #145: libtheoraenc.so.1: cannot open shared object file

Jimbolaya64, 2025-04-27:
When starting the Indi server from Ekos the telescope simulator driver crashes. Log file shows the following error:

libtheoraenc.so.1: cannot open shared object file

Astroarch was updated today (4/27) using "update-astroarch". Please advise. Thank you.

sc74, 2025-04-27:
The libtheora package has been updated to version 1.2. Either wait for Mattia to update indi or compile indi yourself.

To compile, do the following:

Create a directory and copy the file https://github.com/devDucks/astroarch-pkgs/blob/main/packages/libindi/PKGBUILD into it.

Open a terminal in the directory and run the command makepkg -sr PKGBUILD.

Install the compiled package with sudo pacman -U followed by the name of the package ending in .tar.xz

Jimbolaya64, 2025-04-28:
Hmmmm, it appears that the Indi third party drivers are not installed. Is that correct? If so, is there an option to install them? If I must compile them will the github (https://github.com/indilib/indi-3rdparty) instructions work or is there another preferred method? Thank you.

sc74, 2025-04-28:
No, you should have them on your AstroArch. Do a pacman -Qs indi-3rdparty to see what's on your system. You can install them with sudo pacman indi-3rdparty-drivers indi-3rdparty-libs. If your issue is not resolved, you can compile them like libindi

Jimbolaya64, 2025-04-28:
You are absolutely right. The third party drivers there. I was looking for a driver for the SV905C camera which is actually the indi_svbony_ccd driver. My mistake. Thank you again.

MattBlack85, 2025-04-30:
This is considered solved at upstream, I just released Kstars 3.7.6 and Indi 2.1.3 which have been compiled against the newest version of libtheora

closed 2025-04-30 by MattBlack85