devDucks/astroarch #148: Errors running

Riaandw79, 2025-05-01:
Recieve these error during update. Linux noob.

:: Synchronizing package databases...
 astromatto is up to date
 core.db failed to download
 extra.db failed to download
 community.db failed to download
 alarm is up to date
 aur.db failed to download
error: failed retrieving file 'core.db' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed retrieving file 'extra.db' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed retrieving file 'community.db' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
warning: too many errors from mirror.archlinuxarm.org, skipping for the remainder of this transaction
error: failed retrieving file 'aur.db' from mirror.archlinuxarm.org : Operation too slow. Less than 1 bytes/sec transferred the last 10 seconds
error: failed to synchronize all databases (download library error)
resolving dependencies...
looking for conflicting packages...

Packages (1) archlinux-keyring-20250123-1

Total Download Size:   1.18 MiB
Total Installed Size:  1.68 MiB
Net Upgrade Size:      0.01 MiB

Riaandw79, 2025-05-01:
I think Something is wrong. After startup, I did the 

`update-astroarch`

Then 

`sudo timedatectl set-timezone Africa/Johannesburg`

Open Kstars/settings/geography

Enter city, coordinates, elevation and time offset.

Then open Ekos, but it crashes under the simulator or my gear profile.

Riaandw79, 2025-05-01:
Resolved update errors by running from the terminal in astroarch instead of ssh.

Seems to be caused by connection issues.

MattBlack85, 2025-05-02:
Closing this then!

closed 2025-05-02 by MattBlack85