devDucks/astroarch #149: Player One issues after ugrading to INDI 2.1.3

imahgin, 2025-05-01:
Just updated astroarch with latest version -KStars 3.7.6 stable and INDI 2.1.3- and everything seemed to work fine until I tried to take some subs.
If I set the exposure time, it ignores the new setting and the following error is issued:
`[ERROR] Failed to set exposure duration (POA_ERROR_INVALID_CONFIG). `

My camera is a Player One Uranus C. These are the available info in _Indi Control Panel_:
```
Driver Info:
   Name: PlayerOne CCD
   Exec: indi_playerone_ccd
   Version: 1.18
SDK:
   Version: 3.7.1
```

What I've tried so far:

- Remove the PlayerOne CCD Uranus-C_config.xml file: results in the same error.
- Re-compile libindi with 2.1.3 version: results in the same error.
- Manually compile and install  previous INDI version 2.1.2.1: after rebooting _indi_playerone_ccd_ version is the same (1.18) **and now, the driver crashes** 
- Install the original astroarch image: this works flawlessly.
`KStars: 3.7.1
INDI libs: 
PlayerOne driver version: 1.14
SDK: 3.6.2` 
- Use _Native_ connection in AstroDMX: everything works fine, but if I use the INDI connection, then it fails.

According to [this](https://github.com/indilib/indi-3rdparty/releases) link indi-3rdparty-drivers version 2.1.3 was supposed to have the PlayerOneCamera SDK updated to v3.8.0 but in my case it's 3.7.1!!!
I've tried to manually compile 3rd party drivers using the files stored [here](https://github.com/devDucks/astroarch-pkgs/tree/main/packages/indi-3rdparty-drivers) but got compilation errors like these:
```
  /home/astronaut/tmp/3rdparty/src/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:198:52: error: ‘get_max_speed’ was not declared in this scope
  198 |         GTDEConfigurationNP[GT_MAX_SPEED].setValue(get_max_speed(1));
      |                                                    ^~~~~~~~~~~~~
/home/astronaut/tmp/3rdparty/src/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:199:55: error: ‘get_acceleration_angle’ was not declared in this scope; did you mean ‘ahp_gt_set_acceleration_angle’?
  199 |         GTDEConfigurationNP[GT_ACCELERATION].setValue(get_acceleration_angle(1) * 180.0 / M_PI);
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~
      |                                                       ahp_gt_set_acceleration_angle
```

imahgin, 2025-05-01:
As a side note, rolling back to previous versions using _astro-rollback-full_ and rebooting the RBP5 causes crashes in all INDI drivers
As a temporary workaround I've flashed the original astroarch-1.9.0.img in a separate SD.
If further testiing is required I can switch to the 'offending' SD and conduct whatever tests you require.

imahgin, 2025-05-02:
As a side note, I did an astro-rollback-full, but unfortunatelly, after rebooting, EQmod and indi_playerone_ccd drivers crashes.
In the meantime, I flashed another SD with the original astroarch image and it works as expected. The version in that  image are the followings:

- KStars: 3.7.1 Stable
- INDI: 
  - libindi: 2.0.8-2
  - indi-3rdparty-drivers: 2.0.8-1
  - indi-pylibcamera: 2.6.0-5
- PlayerOne CCD Uranus-C
  - Driver name:  indi_playerone_ccd
  - Driver version: 1.14
  - SDK version: **3.6.2**
- AstroDMX -compiled new version-
  - Version 2.14.1
  - Native SDK version for PlayerOne: **3.8.1**

Note that PlayerOne SDK in INDI works fine and is 3.6.2, and Native SDK used in AstroDMX is the latest one (3.8.1) and also works fine.
As stated in the previous comment, the latest astroarch update results in the following INDI versions:

Driver Info:
   Name: PlayerOne CCD
   Exec: indi_playerone_ccd
   Version: 1.18
SDK:
   Version: 3.7.1

I still have the SD with the latest update so feel free to ask if you need me to do any testing on my end.

MattBlack85, 2025-05-02:
@imahgin thanks for the report, this is definitely something with the driver as the problem has been confirmed by others. Gonna have a look at why what I shipped has the 3.7.1 version of the SDK.

imahgin, 2025-05-02:
Looking forward for the fix
If u need me to conduct some testing just ask me.
Thks for your work

MattBlack85, 2025-05-02:
@imahgin I noticed this https://github.com/indilib/indi-3rdparty/commit/84956a621a5298a1ce74fdaa645c703663a10866 which means the the updated has been pushed after the 2.1.3 release. I compiled the bleeding edge Indi as well, can you try to install Indi bleeding and check if the player one works with that?

You can do that with the astro-bleeding-edge command

imahgin, 2025-05-02:
Not sure what do you mean with _astro-bleeding-edge_ command.
How can I install the bleeding edge Indi?
I'm a newbie. Can you elaborate that?

MattBlack85, 2025-05-02:
Launch a terminal (Konsole) and type `use-astro-bleeding-edge`

imahgin, 2025-05-02:
yeap I just saw the command in https://github.com/devDucks/astroarch
sorry
I'm running the command right now

imahgin, 2025-05-02:
I guess I messed up something yesterday when compiling old indi drivers

When running the bleeding command I get an error:
`Errors occurred, no packages were upgraded.`

The log is so huge that I cannot scroll up to the error itself.
Redirecting the console to a file only shows this:

```
:: Synchronizing package databases...
 astromatto downloading...
 core downloading...
 extra downloading...
 community downloading...
 alarm downloading...
 aur downloading...
resolving dependencies...
looking for conflicting packages...

Packages (4) indi-3rdparty-drivers-git-r175.be62be2-1  indi-3rdparty-libs-git-r175.be62be2-1  kstars-git-r176.a10e93d-1  libindi-git-r9974.26f5e7e-1

Total Installed Size:  419.67 MiB
Net Upgrade Size:      391.76 MiB

:: Retrieving packages...
 kstars-git-r176.a10e93d-1-aarch64 downloading...
 indi-3rdparty-libs-git-r175.be62be2-1-aarch64 downloading...
 libindi-git-r9974.26f5e7e-1-aarch64 downloading...
 indi-3rdparty-drivers-git-r175.be62be2-1-aarch64 downloading...
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
Errors occurred, no packages were upgraded.
```
I guess, the best option is to flash the original astroarch image, do an update-astroarch and then test the bleeding edge packages

MattBlack85, 2025-05-02:
@imahgin were you able to compile and install them?

imahgin, 2025-05-02:
Nope, I'm flashing the original astroarch image to start from scratch as KStars also failed to start.
I'll perform the updates in a freshly image and then jump to bleeding edge packages.
Just give me some minutes...

imahgin, 2025-05-02:
@MattBlack85 I finally managed to fix ths issue by jumping to the beta packages as you suggested!!
When the new version is available, do I need to switch to the stable flavour and then update astroarch?
Thk u very much!!!!

I think you can now close this issue

imahgin, 2025-05-02:
[image]

imahgin, 2025-05-02:
[image]