devDucks/astroarch #150: Having trouble installing indi-asi-power

Riaandw79, 2025-05-02:
Hi

tried all the usual methods. sudo apt-get fails. apt install indi-asi-power fails.
Any advice?

ASAI-power is available to select in ekos but I cannot start it in the indi server gui

Riaandw79, 2025-05-02:
[image]

sc74, 2025-05-02:
The apt command you used is not an ArchLinux command but an Ubuntu one. The indiserver GUI utility does not list all the drivers like in the ekos panel. Are you using ekos to start the indi server with the selected drivers? When you selected asi power from the auxiliaries and launched ekos, did you get an error?

MattBlack85, 2025-05-02:
@Riaandw79 will have a look at the indiserver UI code and let you know. Possibly there is a bug in my program

Riaandw79, 2025-05-02:
> The apt command you used is not an ArchLinux command but an Ubuntu one. The indiserver GUI utility does not list all the drivers like in the ekos panel. Are you using ekos to start the indi server [...]

Thank you. That makes sense why none of my usual inputs worked. 

When I start Ekos step 2 it just hangs there, never connects.

Riaandw79, 2025-05-02:
> [@Riaandw79](https://github.com/Riaandw79) will have a look at the indiserver UI code and let you know. Possibly there is a bug in my program

Awesome. Thanks for the support. Everything else seems smooth and stable.

I activated the 12v hub in config.txt for the meantime.

MattBlack85, 2025-05-02:
Alright, can reproduce this. It seems the ASI power driver is not built, so it's not on the list cause there is no ASI power driver on the system. Not sure why the build doesn't show a failure but it seems the error is passed silently. That driver requires pigpio, I will look if it's possible to build it as I vaguely remember I had some issues with pigpio in the past

MattBlack85, 2025-05-05:
@Riaandw79 I was able to compile the driver, it should appear in the bleeding edge packages later this week