devDucks/astroarch #152: System Monitor with RPi5

HLex79, 2025-05-04:
Hello,

I wonder about the sensors nor other values being shown in system monitor? I wonder as the fan of my rpi5 is never noticeable...

[image]

Would be nice to have those functions aboard :-)

BR, 

Lex

MattBlack85, 2025-05-05:
Hey @HLex79 as far as I know this problem haunted AstroArch since day 0. For some reasons it seems the kernel is not compiled with the necessary things needed for that. Just out of curiosity, what happens if you click on fix?

sc74, 2025-05-05:
The button doesn't fix anything because the plasma-ksystemstats.service service isn't working on Arm Arch. Perhaps a ticket should be opened for this issue at https://invent.kde.org/plasma/ksystemstats/-/issues?

In the meantime, it's possible to get the fan rotation value by installing the psensor package

closed 2025-12-15 by MattBlack85