devDucks/astroarch #153: Raspberry pi camera v3 not working

Dupie-123, 2025-05-29:
Hi

I have done a fresh install using v 1.9.0. Ran update-astroarch and then after a reboot I ran sudo pacman -S indi-pylibcamera. When I run the command libcamera-hello -t 0 I get the following output:

libcamera-hello: error while loading shared libraries: libboost_program_options.so.1.87.0: cannot open shared object file: No such file or directory

I have tested this with and without making changes to config.txt ie. 

camera_auto_detect=0
dtoverlay=imx708

I have already confirmed that the camera is functioning on a raspbian install with the libcamera library. Can you please assist me in getting the camera to operate with INDI

Thank you

sc74, 2025-06-04:
The libcamera package needs to be modified due to updates. Testing is underway

MattBlack85, 2025-09-03:
@Dupie-123 is the camera working after the latest update?