devDucks/astroarch #159: After update-astroarch SDDM forces to use wayland (no possibility to switch to plasma X11)

gdq, 2025-06-21:
Not sure if this is a astroarch related issue or rather a more general arch linux issue, but after having run update-astroarch SDDM no longer uses plasma X11, but wayland.
I don't mind using wayland, but I can no longer remote login with (Real)VNC.
Tried anydesk and rustdesk but that did not work out either.
Is this an issue or do I have to settle with wayland?

xcentricdiff, 2025-06-21:
pacman -S kwin-x11

https://archlinux.org/news/plasma-640-will-need-manual-intervention-if-you-are-on-x11/

thekgb09, 2025-06-22:
Still fighting this on a Raspberry Pi 5. Adding plasma-x11-session and kwin-x11 and changing the session type did not fix the issue after multiple reboots.
Trying to figure this out on a clear night was not the plan.

xcentricdiff, 2025-06-22:
I just did another `update-astroarch` to take care of the `linux-firmware` issue. Installed without issue. Then I ran `pacman -Syu plasma-x11-session` which also updated `kwin-x11`. I checked the session type in `/etc/sddm.conf.d/kde_settings.conf` was `plasma-x11`.

I can confirm @thekgb09.  VNC login fails. (Rpi4)
- It presents with the `sddm` greeter
- it takes the `astronaut` password
- throws a black screen
- and disconnects.

xcentricdiff, 2025-06-22:
Good ol' `journalctl -u sddm` announced: "Unable to find autologin session entry "plasma-x11"
I changed it back to `plasmax11` and everything works.

(I swear it needed to be `plasma-x11` in order to work yesterday, before the `linux-firmware` updates.)

Thanks folks for the help!!

imahgin, 2025-06-22:
By the way, are you using the hotspot to connect to the RBP?
My screen freezes after some seconds of inactivity and I can recover if that time is lower than 10-30 secs. If exceeded, I need to close and open again TigerVNC/RealVNC.

Using the wired connection works without any problem.
Using a wireless bridge device to connect to the hotspot also works -but with some extra delays, of course-
Wireless power save is off

Any hint?

gdq, 2025-06-23:
VNC was problematic via cable connection as well as hotspot.
That issue was solved after installing plasma-x11-session and kwin-x11
But anyhow ran into other issues after that as I did not go thru the manual firmware update.
So back to creating a new image, update-astroarch, manual firmware update, installing plasma-x11-session and kwin-x11
And configuring indi with my equipment.
Luckely this is a "hobby" project 

;-)

sc74, 2025-06-23:
@gdq It should be remembered that AstroArch is based on ArchLinux which is a rolling distribution ;)

gdq, 2025-06-23:
> VNC was problematic via cable connection as well as hotspot. That issue was solved after installing plasma-x11-session and kwin-x11 But anyhow ran into other issues after that as I did not go thru [...]

Lesson learned: before running update-astroarch check https://archlinux.org/ :)

MattBlack85, 2025-07-10:
@sc74 I got another friend having this issue which solved by installing plasma-x11-session, is this package required now?

PR #163 (2025-07-18): Release 2.0
What's new:

- Fixed HDMI issue
- Fixed plasma x11 session issue
- Added visual tool to change settings
- Fixed xorg.conf config

Fix #159 
Fix #161
Fix #155
Fix #133
Fix #129
Fix #120
closed 2025-07-18 by MattBlack85