devDucks/astroarch #162: Updated today, KStars fails to start

picciux, 2025-07-04:
Hi all,

after upgrading a few hours ago, KStars fails to start complaining that libopencv_core.so.411 doesn't exist. In fact, opencv 4.12 is installed.
As a side note, also Chromium doesn't start cause not found lib (actually searching for a previous version lib), and udisksd the very same (but different lib).

Any ideas?
Thank you very much.

Matteo

Clay2F, 2025-07-05:
I am not so familiar with Linux based systems. Hoe can I start kstars during the terminal?

sc74, 2025-07-05:
For those who would like to downgrade the OpenCV package to use kstars while waiting for Mattia's update.

Install the downgrade package

paru -S downgrade

Then run sudo downgrade opencv and choose the previous version

MattBlack85, 2025-07-05:
I tried to recompile and I got some errors, I rebooted my pi remotely and it seems it doesn't come up anymore. I will come back tomorrow evening, will look into it after that.

For the time being downgrade the opencv package

picciux, 2025-07-05:
> For those who would like to downgrade the OpenCV package to use kstars while waiting for Mattia's update.
> 
> Install the downgrade package
> 
> paru -S downgrade
> 
> Then run sudo downgrade opencv and choose the previous version

Someone could borrow me an opencv 4.11 package? I don't have it in package cache, so I cannot downgrade...
tks

JJGodoy5, 2025-07-06:
> > Version 4.11.0.13 of opencv https://www.swisstransfer.com/d/450e5d70-81cc-4cfc-86fc-82f420cd10f0
> 
> Thankyou very very much, it worked :)

Could you please explain how you solved everything using the link?
I've tried several ways, but I feel like I don't know enough about Linux.

sc74, 2025-07-06:
Install the downgrade package with the command:

paru -S downgrade

Use downgrade to revert to a previous version of onpencv 4.12 with the command:

sudo downgrade opencv

Choose version 4.11

If downgrade doesn't find a previous version of opencv 4.12 on your PI, download the files and copy them to the pacman cache folder with the commands:

sudo cp opencv-4.11.0-13-aarch64.pkg.tar.xz  /var/cache/pacman/pkg/

sudo cp opencv-4.11.0-13-aarch64.pkg.tar.xz.sig  /var/cache/pacman/pkg/

Then run downgrade again to choose version 4.11

JJGodoy5, 2025-07-06:
It worked!
Thank you very much.
Chromium still doesn't work, but the really important thing is that Kstars does. When I chose to install the OpenCV pre-package, it asked me a question, to which I answered yes. Maybe that's the reason Chromium isn't working.

I'm satisfied so far.

MattBlack85, 2025-07-07:
Kstars 3.7.7-3 has been now released, the system can be upgraded fully again

closed 2025-07-07 by MattBlack85