devDucks/astroarch #166: Astroarch died after update... pi5

miguegh, 2025-07-24:
Hi all... sad but true ( I wish it was the Metallica´s song).. I have been working on the electronic to controll a roll off observatory with Arduino and Rolloffino driver.. Electronic ready to test.. Switched on mi pi5.. update warning about linux-firmware.. ok, I executed the suggested command.. 
"You have not administrative right to process this.." after do it.. (using "astro" like always on this distro.. 
Well.. I executed an update-astroarch... some minutes before.. everything looks fine... some new on the desktop "Astroarch Tweak Tool" or so... 
I tried one more time to execute the suggested command "pacman -Rdd linux-firmware &&pacman -Syu linux-firmware"
"linux-firmware" not found.. Strange but I launched Kstars.. launched Ekos.. create a new profile to test Rolloffino.. and shutdown to move me to the workshop where I have the electronic and plug the Arduino to test... 
After some minutes not hotspot to connect to it... the fun on the pi5 is running but not flashing led... looks death.. power off.. come back to my desktop PC, plug the pi5 to the network (wired), power on... no accessible by network.. no by vnc, no by Filezilla... not reachable... 

I understand is an issue with the linux-firmware... during de installation I could read something like a problem with amd64 architecture.. 
I can flash a new card and made everything again but, once I updated it I expect the same result.. 
Any advice?? 
Thanks

miguegh, 2025-07-24:
Thanks @MattBlack85 no.. I only executer Kstars, create a new profile on Ekos to include the Rolloffino, quit, shutdown and move myself with the pi5 to the workshop. Plugged the Arduino  and the usb to serial for the telescope controller and power on the pi5 with the explained result.. I´m flashing the card now.. and try to setup all tomorrow .. 
If you confirm that this is not an update issue I will put focus on things around... thanks

MattBlack85, 2025-07-24:
I will seek and destroy any potential external factor. I did the update around 1 week ago, something may have changed but hard to say at the moment

imahgin, 2025-07-25:
Use ssh and then execute this:

> sudo pacman -Syu plasma-x11-session

Then reboot

miguegh, 2025-07-29:
Hi All... I have flashed again the card and now everything is working.. I´m still receiving the "issue alert" about the update.. but when I try to execute the suggested command "pacman -Rdd linux-firmware &&pacman -Syu linux-firmware" the result is the same: 
"linux-firmware" not found. But for this moment and like everything looks fine.. I will follow with my test of Rolloffino driver.. at this moment I´m receiving a "can connect" response to it.. but this is another fight..

miguegh, 2025-07-29:
hi @MattBlack85 thanks.. so I will ignore the message.. I will try to solve now my problem with the Arduino and Rolloffino driver about why it is not connected with it..

miguegh, 2025-07-30:
Hi All... Rolloffino is working correctly... my problem of non connection was caused between the Arduino and one prolific USB to Serial converter used to control a SkySensor 2000 PC.. I´m not sure but it looks if the USB port on the pi5 were specially sensitive and prone to mix peripherals that generate serial ports.. 
Now Rolloffino is working and also the SkySensor... but I have not full control of the SkySensor 2000 PC.. For example: I can control it from Kstars (via Ekos/INDI) and ask to goto to a star, once its reache I can "sync" and the SS2k asume it as alignment point. Great.. but when I try to move the motors from the mount motion control on EKOS to for example center the object.. nothing happens.. I can not control from this window.. no response... I know.. this could be in another issue..  I will follow testing

closed 2025-07-31 by miguegh