devDucks/astroarch #175: Openbox window manager. Feature request.

bgunn9in, 2025-08-30:
Please add support for the Openbox window manager for Raspberry Pi 4 users. It consumes significantly fewer system resources. Let Openbox be optional, with KDE remaining the default, and allow the user to switch to it if they wish. I have already tried installing Openbox in AstroArch myself and achieved some positive results. However, my knowledge of Linux does not allow me to fully integrate it.

IngarKCT, 2025-08-31:
It might be worth looking into LXQT as a lightweight KDE alternative. It runs fine on a Pi4 with sdcard and on vanilla arch it uses openbox by default.

MattBlack85, 2025-08-31:
hi people! I'd like to start with this - I am a big fan of WMs, I use a tiling WM on my everyday Arch x64 box (Xmonad) and I am experimenting with hyprland, WMs are cool!

The problem for something like AstroArch is the configuration, the system as you see it now is heavily configured to make it working out of the box with something that will make happy(ish) the majority of the users, additionally we store all of that configuration in the repo itself and we mantain it.
Any additional DE or WM must bring in a decent starting config for the end user that works and must be kept up to date through updates (the WM itself and the system). 
As you can see at the moment it's just 2 people mantaining astroarch and this takes already a decent chunk of our time, additional configs will require more time (if done by us, as we don't know openbox) so unless there is someone who will volunteer to make a starting config that works and will keep it up to date, the chance to see new DE/WM is close to 0 :(

Happy to chat if any of you want to tackle this goal!

MattBlack85, 2025-08-31:
DE is a desktop environment, like KDE or gnome are a full GUI suite which contain a windows manager (how windows appear and behave), a bar, a panel, a system tray, file manager etc. It's like when you go to the car dealer and you buy a car.

A WM is a windows manager who tells the system how windows should be rendered, how they move, they resize etc. It's like going to the car dealer and buy the steering wheels and the pedals only. It's usually extremely light and they look astonishing. They are controlled exclusively by keyboard shortcuts and by default have nothing else, you build on top of them adding a bar and other stuff you may find useful for your system

FireNWater, 2025-08-31:
OK, thanks. I think this would add more moving parts to AA at this time. Perhaps later?