devDucks/astroarch #176: download iso for rpb5

juanmatiasb-alt, 2025-09-02:
Good morning everyone, how are you? I'm bothering you to ask how I download Astroarch, because when I download it, it only downloads 3 MB... Thank you very much.

[image]

[image]

[image]

MattBlack85, 2025-09-03:
hi @juanmatiasb-alt, downloading the zip will download the source code form this repository, check the README now, I put back the link to the old image

MattBlack85, 2025-09-04:
solved

closed 2025-09-04 by MattBlack85