devDucks/astroarch #177: Update v1.9 image, update-astroarch failure on conflicting files

juanmatiasb-alt, 2025-09-04:
Good morning everyone, how are you?

I comment on my problem

Thank you very much in advance.

> Raspberry pi 5 with astroarch 1.9

> After initial boot, connected to local wifi and ran 'update-astroarch'. Update failed with:

[image]

juanmatiasb-alt, 2025-09-05:
hi @MattBlack85 

Thank you very much for your time

[image]

MattBlack85, 2025-09-05:
and you need to keep the other packages in the command as well

juanmatiasb-alt, 2025-09-05:
@MattBlack85 

Something like that? I don't think I'm typing correctly at the moment.

[image]

juanmatiasb-alt, 2025-09-05:
hi @MattBlack85 

I think it works better this way... without "kwin"... I'll continue with the next one "xdg-desktop-portal-kde"

[image]

juanmatiasb-alt, 2025-09-07:
I think the last clean install was a success!

Is that true?

Thanks for commenting.

[image]

MattBlack85, 2025-09-08:
@juanmatiasb-alt that seems the case, what was the difference compared with the latest trial?